Seithan/EasyNextionLibrary #1: GIVE THIS ERROR

axelpablo, 2020-04-12:
libraries\EasyNextionLibrary\EasyNextionLibrary.cpp.o (symbol from plugin): In function `EasyNex::callTriggerFunction()':

(.text+0x0): multiple definition of `EasyNex::callTriggerFunction()'

sketch\ARFTHERM.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1

Seithan, 2020-04-12:
Can you please tell me how you declare and how you call the trigger() function if you can?
Could you please send me the .ino file sketch you are using, if it is ok with you.

axelpablo, 2020-04-12:
#include <callTriggers.h>
#include <EasyNextionLibrary.h>
#include <EEPROM.h>
#include <Wire.h>
#include <RTClib.h>
#include <OneWire.h>
#include <DallasTemperature.h>

EasyNex myNex(Serial);

// Data wire is plugged into digital pin 2 on the Arduino
#define ONE_WIRE_BUS 2 //PIN DOVE SENSORE DS18B20 E' COLLEGATO
#define DS1307_ADDRESS 0x68  // works for DS3231 unchanged
RTC_DS1307 rtc;

byte zero = 0x00; //workaround for issue #527
// Setup a oneWire instance to communicate with any OneWire device
OneWire oneWire(ONE_WIRE_BUS);  

//char buffer[100] = {0};

// Pass oneWire reference to DallasTemperature library
DallasTemperature sensors(&oneWire);

char buffer[100] = {0};
//double tL=25;

//DICHIARAZIONE COMPONENTI NEL NEXTION

void setup(void)
{
  myNex.begin(9600);
    sensors.begin(); 
  Wire.begin();

 if (! rtc.isrunning()) { 

setDateTime();

 }

}

void Temperaturasondatonextion(){

  float t = sensors.getTempCByIndex(0);
   static char TemperaturaAttuale[5];
dtostrf(t, 4, 1, TemperaturaAttuale);

}

void printDate(){

  // Reset the register pointer
  Wire.beginTransmission(DS1307_ADDRESS);
  Wire.write(zero);
  Wire.endTransmission();

  Wire.requestFrom(DS1307_ADDRESS, 7);

  int second = bcdToDec(Wire.read());
  int minute = bcdToDec(Wire.read());
  int hour = bcdToDec(Wire.read() & 0b111111); //24 hour time
  int weekDay = bcdToDec(Wire.read()); //0-6 -> sunday - Saturday
  int monthDay = bcdToDec(Wire.read());
  int month = bcdToDec(Wire.read());
  int year = bcdToDec(Wire.read());

  //print the date EG   3/1/11 23:59:59

char orario[30]; //
sprintf( orario, "%02hhu:%02hhu", hour, minute );

 }
 

void setDateTime(){

  byte second =      00; //0-59
  byte minute =      01; //0-59
  byte hour =        23; //0-23
  byte weekDay =     3; //1-7  Monday is 1!
  byte monthDay =    07; //1-31
  byte month =       04; //1-12
  byte year  =       20; //0-99

  Wire.beginTransmission(DS1307_ADDRESS);
  Wire.write(zero); //stop Oscillator

  Wire.write(decToBcd(second));
  Wire.write(decToBcd(minute));
  Wire.write(decToBcd(hour));
  Wire.write(decToBcd(weekDay));
  Wire.write(decToBcd(monthDay));
  Wire.write(decToBcd(month));
  Wire.write(decToBcd(year));

  Wire.write(zero); //start

  Wire.endTransmission();

}

byte decToBcd(byte val){
// Convert normal decimal numbers to binary coded decimal
  return ( (val/10*16) + (val%10) );
}

byte bcdToDec(byte val)  {
// Convert binary coded decimal to normal decimal numbers
  return ( (val/16*10) + (val%16) );
}
//--------------------------------------------

void trigger2(){
//myNex.writeStr("t0.txt", "P");
  
}
void trigger3(){
myNex.writeStr("tp0Temp.txt", "H");

}

//FUNZIONE LOOP-RIPETE

void loop(void)
{ 
    myNex.NextionListen();
  
  //delay(1000);
  //sensors.requestTemperatures(); 
  
 //printDate();
// Temperaturasondatonextion();
 

 
}

axelpablo, 2020-04-12:
I have try to remove lines 16,17,18 in EasyNextionLibrary.cpp and it seems work...

axelpablo, 2020-04-12:
Now,othre problem...
In the nextion graphic,there are two pages.
Page0 and page1.
With one picture(button use) for page.

page0 --> Picture click(printh 23 02 54 02):
void trigger2(){
//myNex.writeStr("t0.txt", "P");(t0.txt in in same page)
  
}
AND NO WORK

page1 --> Picture click(printh 23 02 54 03):
void trigger3(){
myNex.writeStr("tp0Temp.txt", "H");(tp0Temp.txt NOT IN SAME PAGE)

}
AND WORK!

CAN YOU HELP ME ?

Seithan, 2020-04-12:
#include <callTriggers.h>
You do not have to call the callTriggers.h The only thing you must include in your sketch, is the:
`#include "EasyNextionLibrary.h"`
Instead of <> use ""

Seithan, 2020-04-12:
> I have try to remove lines 16,17,18 in EasyNextionLibrary.cpp and it seems work...

Library calls the` callTriggers.h` from there in order to not have to include it on the sketch code:
if you remove those lines then you must include the  `callTriggers.h` in every sketch code.
The writeStr() command writes only on the currently loaded page. It cannot write at another page even if you had the txt global, you must call it with the page Name in front of txt.
Give me some time to check your whole sketch code. I would suggest you used the library as it was, with no modifications, as it has been tested multiple times

Seithan, 2020-04-12:
````Cpp
void trigger2(){
//myNex.writeStr("t0.txt", "P");(t0.txt in in same page)
}
````
As I saw, the command is correctly written, which means that the t0.txt will become "P".
I guess that the comment was not in the code you are using. 

What you also must check is that the send component id is not checked, because if they are, they are going to send unwanted commands over Serial.

`myNex.writeStr("tp0Temp.txt", "H");(tp0Temp.txt NOT IN SAME PAGE)`
With this you mean that when you are on page1, the command is executed?
OR that you are on another page and the command changes the tp0Temp textbox that is on another page?

Please give me a feedback as soon as possible.

axelpablo, 2020-04-13:
Yes,in the code i no have comment.(//myNex.writeStr("t0.txt", "P");(t0.txt in in same page)).
What you mean "send component id is not checked"? How i can check?

I did some tests, and "myNex.writeStr (" tp0Temp.txt "," H ");" it works only if the page with the "tp0Temp.txt" component is open.
I solved it by doing this:

void trigger3 () {
myNex.writeStr ("page page0");
myNex.writeStr ("tp0Temp.txt", "H");
}

First I go to the page where the component is, and then I change the text.
Before this I also tried putting the txt in global, but it didn't work.

Seithan, 2020-04-13:
Well as I wrote you above:

> The writeStr() command writes only on the currently loaded page. It
> cannot write at another page even if you had the txt global, you must
> call it with the page Name in front of txt.
I m going to make more clear at the documentation to avoid misunderstanding.
**To write in a global component**  you must refer the `page name`  (no page ID) before the components name like this
````Cpp
myNex.writeStr("page0.t0.txt", "HELLO") // same for Number
myNex.writeNum("page0.n1.val", number);
````
if you don't need something else for this issue you can close it.
Thank you

closed 2020-04-13 by axelpablo