Seithan/EasyNextionLibrary #2: myNex.readNumber()

axelpablo, 2020-04-13:
In my case i have a Variable in nextion(ntu).
ntu.val=489
In Arduino code:
uint32_t number = 0;
number = myNex.readNumber("ntu.val");
Serial.println(number); 

But the output(Serial.println(number);) is "4294967066".
Not the real val(489).

Seithan, 2020-04-13:
Check that the number component that you ask for its value is on the current loaded page.
you can see at **ChangePagesAndSendFloatValues** example how you can use the `myNex.currentPageId`
and keep the sketch code synchronized with Nextion page. 
For get the value of a numeric component from a page that is not loaded you must turn it to global scope and access it with a refer to page name: 
````Cpp
number = myNex.readNumber("pageName.n0.val");
````  
The folowing is from the [Nextion Editor Guide](https://nextion.tech/editor_guide/)

> In an HMI project a page is a localized unit. When changing pages, the
> existing page is removed from memory and the requested page is then
> loaded into memory. As such components with a variable scope of local
> are only accessible while the page they are in is currently loaded.
> Components within a page that have a variable scope of global are
> accessible by prefixing the page name to the global component
> .objname.
> 
> As an Example: A global Number component n0 on page1 is accessed by
> page1.n0. A local Number component n0 on page1 can be accessed by
> page1.n0 or n0, but there is little sense to try access a local
> component if the page is not loaded. Only the component attributes of
> a global component are kept in memory. Event code is never global in
> nature.

Seithan, 2020-04-13:
I made some changes to the .cpp file in order to have a return of 7777 at readNumber() function and for the case that Nextion hasn't sent the data we want. This is going to cover the case of asking to read a number from a page that is not loaded. You can try it and see if it covers your needs. 
Thank you for opening an issue and letting me know something that I may not had noticed.
If you do not need anything else for this issue, please close it.

axelpablo, 2020-04-13:
Thanks
I have change .ccp file with the new version.
In nextion,is open page2,and with pciture  in this page i send command("printh 23 02 54 03")
In same page i have "Number" component(ntu.val),with value is "574"
In the Arduino code:
void trigger3(){
uint32_t number = 0;
number = myNex.readNumber("page2.ntu.val");   
       Serial.println(number);
}
But the output is "page2.ntu.val⸮⸮⸮4294967066".
(But the real value in nextion is "574".

axelpablo, 2020-04-13:
And all component are set to global.(ntu,picture,...)

axelpablo, 2020-04-13:
I have try to change val of ntu.
But the output in Serial.println,is always "print page2.ntu.val⸮⸮⸮4294967066".

axelpablo, 2020-04-13:
I have remove from Touch Release Event of picture(use for send command),the code: page page0.
And now it work ("myNex.readNumber"page2.nut.val")!

But always first time it give me the real value.
In the next "myNex.readNumber"page2.nut.val" it give me always 7777.
This the output:
page page1⸮⸮⸮page page2⸮⸮⸮printh 23 05 52⸮⸮⸮print page2.ntu.val⸮⸮⸮574
printh 23 05 52⸮⸮⸮print page2.ntu.val⸮⸮⸮7777
printh 23 05 52⸮⸮⸮print page2.ntu.val⸮⸮⸮7777
printh 23 05 52⸮⸮⸮print page2.ntu.val⸮⸮⸮7777
printh 23 05 52⸮⸮⸮print page2.ntu.val⸮⸮⸮7777

Seithan, 2020-04-13:
Sorry that I miss it before.
As I can see you configure the Library to use the Serial ( Serial):
````Cpp
EasyNex myNex(Serial);
````
and to the trigger3() function
````Cpp
void trigger3(){
uint32_t number = 0;
number = myNex.readNumber("page2.ntu.val");
Serial.println(number);
}
````
you call the Serial and you are sending a println for the number variable, I'm guessing that the Serial is connected on Nextion and if I'm guessing right then you send to Nextion an unknown command.
If you want to check, for debug reasons, that the value of `ntu.val ` is the right one and that there is no problem with reading the components value :
Create a new temporary numeric component on the page that is loaded. Supposing that the name of the new numeric is `n22` then format the trigger3() fuction like this:
````Cpp
void trigger3(){
uint32_t number = 0;
number = myNex.readNumber("page2.ntu.val");
//Serial.println(number);
myNex.writeNum("n22.val", number);
}
````
Now you are going to see the number that is listed on `ntu`

axelpablo, 2020-04-13:
Thanks for your help.
I've solved with this code:

```C++
uint32_t number = 0;
uint32_t lastnumber = 0;
void trigger3(){
number = myNex.readNumber("page2.ntu.val"); 
if(number != 7777){       // 7777 is the return value if the code fails to read the new value
  lastnumber = number;
} else if(number == 7777){
    number = lastnumber;
}
   Serial.println(number);  
}
```
You think is correct with this?
It work.

Seithan, 2020-04-13:
This is a way to confirm that the reading of the the value doesn't fail.
Try the code that I was sent you before to see the value on the display and tell me if you have the right value.
Can you please tell me where are you sending the number with the Serial.println(number); ?
and where you are viewing it ?

axelpablo, 2020-04-13:
> Can you please tell me where are you sending the number with the Serial.println(number); ?
> and where you are viewing it ?
In serial monitor,in Arduino IDE.

axelpablo, 2020-04-13:
Is possible to read a text from Text component in Nextion ?

Seithan, 2020-04-13:
No, I haven't add this feature as I thing that the usage is almost unnecessary as a Display is most for display message and not to sent text to MCU, I tried hard to put as less functions as possible.
The profit of adding this function is small in front of keep the simplicity.

Seithan, 2020-04-13:
but if someone need to send text, he can do it, using the custom protocol

axelpablo, 2020-04-13:
okok,thanks.
If i want print "number" in Text component in Nextion,is possible convert it ?
I see "number" is a uint32_t.

And not possible with: 
myNex.writeStr("tTEMPSET.txt", number);

Seithan, 2020-04-13:
This is from the example ChangePagesAndSendFloatValues it is for float but uint is the same
````Cpp
String tempString = String(temperature, 1); // Convert the float value to String, in order to send it to t0 textbox on Nextion
    myNex.writeStr("t0.txt", tempString);       //Write the String value to t0 Textbox component
````

Seithan, 2020-04-13:
Im going to add a case for this, on the write text example

axelpablo, 2020-04-13:
Thanks,it work.
Thanks for the help and congratulations for this library.

closed 2020-04-13 by axelpablo