Seithan/EasyNextionLibrary #3: Cannot use software serail

wesselah, 2020-05-12:


wesselah, 2020-05-12:
When compiling on a arduino micro i do get this error 

no matching function for call to 'EasyNex::EasyNex(SoftwareSerial&)'

I am wondering do i have to change this line in easyNextionLibrary.h
EasyNex(HardwareSerial& serial); ???
any advice would be appriciated

Seithan, 2020-05-12:
Hello,
I did not want to put Software Serial feature in the library, in order to keep it as simple as possible. 
And I believe, because the display is the main device, it has to be to the Hardware Serial ports.
If you need an extra Serial for another device, you can set a Software Serial for the other device. Not for Nextion.
It is not that simple to change with one line. See if you can keep Nextion at Hardware Serial and the other device you need at the Software Serial.
If you still need to use Nextion at the Software Serial, give me some time to send you a modification.

Seithan, 2020-05-12:
Hello,
As I finished the modifications I promised, I will attach the new EasyNextionLibrary.cpp and EasyNextionLibrary.h files that you must replace, in order to have Software Serial. Pay attention. It is only for Software Serial. Not Hardware.
I also have the Four Step Example modified for Software Serial.
[FourStepExampleSoftwareSerial.zip](https://github.com/Seithan/EasyNextionLibrary/files/4618422/FourStepExampleSoftwareSerial.zip)

Have a nice evening! If you do not need something else, you can close this issue.

wesselah, 2020-05-13:
Thanks that was fast
I wil test and see if software serial works.
Could not use hardware serial because i am using a arduino with a 32u4 and i am using the hardware serial as keyboard to the pc so i need that part.
I will test and let you know

wesselah, 2020-05-13:
Works like a charm thx

closed 2020-05-13 by wesselah