Seithan/EasyNextionLibrary #4: Read Variables

Sinix10, 2020-08-12:
Hello,

I don´t see a function to read/write Nextion´s variables.
Is there a way to make it work?
Thanks

Seithan, 2020-08-12:
Hello Sinix10,

If I understand correctly, you want to read/write the **sys** variables or any other variables you have added in the .HMI

You can read/write the variables as any other component.

Use readNumber() to read the value of a numeric variable.
Example: myNex.readNumber("va0.val");
**BUT:** myNex.readNumber("sys0");

Use writeNum() to change the value of a numeric variable.
Example: myNex.writeNum("va0.val", 255);
**BUT:** myNex.readNumber("sys0", 375);

Use readStr() to read the text of a String variable.
Example: myNex.readStr("va0.txt");

Use writeStr() to change the text of a String variable.
Example: myNex.writeStr("va0.txt", "Hello World");

For this to happen, the variables you want to read/write must be at the page you are currently on.
Otherwise, if the variables are of **global** scope, you will need to use a prefix with the page name that the variables are at.
Examples:
myNex.readNumber("page0.va0.val"); // If the variable is at page0
The same goes for the other functions as well.

I hope that I have answered your question.
If you do not have further questions on this subject, please close this issue.

Sinix10, 2020-08-13:
Yes I´m referring to Variable objects, AKA type 52 in the Nextion Editor.
I confirm that, the function myNex.readNumber(), does in fact work with Nextion Variables, but you did not inform it on the OP, please add it, as it is a very useful information. (you only refered number object aka type 54, text object aka 115 and several buttons object, but never Variables object. )
Also the "myNex.readNumber("page0.va0.val"); // If the variable is at page0" is not present on the OP, please add it, as it is also  a very good information.
Thanks a lot mate!

Seithan, 2020-08-13:
OK Sinix10
I will make it more clear where it is needed as soon as possible.
Thank you for the comment

closed 2020-08-13 by Seithan