Seithan/EasyNextionLibrary #5: How to read page name

Skorpi08, 2020-08-19:
Now i need the printh Line for reading that the page was changed.
How can i do it without printh 23 02 50 01 ?

void loop(){
  myNex.NextionListen();                          
 if(myNex.currentPageId != myNex.lastCurrentPageId){ 
  newPageLoaded = true;                                     
    currentpage = myNex.readStr("pages.txt");
    myNex.writeStr("t1.txt",currentpage);
    newPageLoaded = false; 
    myNex.lastCurrentPageId = myNex.currentPageId;
     lastpage = currentpage;
    }                                   
}

[image]

Seithan, 2020-08-19:
Hello Skorpi08,

I am sorry but I did not fully understand the question. Maybe you should tell me what exactly is the problem you are facing.

The library uses that specific protocol to identify the ID of the pages. Sure, there are many other ways, such as using the **dp** variable.

I am waiting for your reply :)

Skorpi08, 2020-08-19:
Sorry for my bad English :)

With library i can read only pageID, but i want read pageNAME.
With printh 23 02 50 01  i trigger that the page was changed and then i read the pages.txt with get command.
So the idea is read pages.txt without polling every second or so without the printh command.

Is it possible add this "feature" to your protocoll?
Send with print "#4Ppage0"

Seithan, 2020-08-19:
Usually, all the commands coming from Nextion, like **dp** or **sendme**, refer to the pageID.
I understand that you want to display the name of the page and not the ID. 

EasyNextionLibrary has an option that you can modify it however you want, at cases 'X' or 'Y' at the **readCustomCommands.cpp**.

In my opinion, I would not do it that complicated. Provided that the library saves every time you change page the pageID to the  **currentPageId** variable as long as you have the **printh** command at the Preinitialize event of every page **printh 23 02 50 pageID**. So, you can use the **currentPageId** with a **switch() case** function or multiple **if()** and **else if()** functions and assign the pageID to a name.
Example: 

if(myNex.currentPageId == 1){
  **String** pageName = "page1";
} else if(myNex.currentPageId == 2){
  **String** pageName = "page2";
}

or with **switch() case**

switch(myNex.currentPageId){
  case 1:
    **String** pageName = "page1";
    break;

  case 2:
    **String** pageName = "page2";
    break;
}

Then use pageName variable as you like. You can send it to a t0 textbox if you like with myNex.writeStr("t0.txt", pageName); 

Do not hesitate to ask if you have further questions :)

closed 2020-08-19 by Skorpi08