Seithan/EasyNextionLibrary #8: readNumber() not working with SoftwareSerial

Jawomo, 2020-09-22:
As SoftwareSerial is more felxible and doesn't block Serial for debugging i changed the library to use SoftwareSerial.

`EasyNex::EasyNex(SoftwareSerial& serial)`

I get the trigger1 events as well i can write numbers or strings. But readNumber() is not working at all. Always returns 777777

Already tried playing with the timeouts, but still no luck.

Would you add support for SoftwareSerial or help to find the cause of readNumber() not working? 
Appreciate your hard work!

Seithan, 2020-09-22:
Hello,

I usually prefer to use the Hardware Serial, as it is more reliable and Nextion or any other display is the most critical component in a project and I use the Software Serial for the other components, which might want to use Serial.
This, however, is something that depends on how one sees it.

If you want to use the library with Software Serial, I have a version with Software Serial and I have published it in older issues that Software Serial was needed by different users.

I will attach the .zip file with the .h and .cpp files, which need to be replaced for version 1.0.4.
I have not made this a feature of the library, because I will need to make the function to run simultaneously with Software and Hardware Serial.
To do that, I will need to use the Stream class and by doing that, it would not follow the [APIStyleGuide](https://www.arduino.cc/en/Reference/StyleGuide) of Arduino, which has a primary scope of keeping things as simple as possible, in order for everyone to understand. Especially the beginners. 

[1_0_4_SoftwareSerial.zip](https://github.com/Seithan/EasyNextionLibrary/files/5263377/1_0_4_SoftwareSerial.zip)

It is tested and it is working.

If you do not have further questions, please close this issue :)

Jawomo, 2020-09-23:
readNumber() is still not working. Everything else is working like readStr, writeStr, writeNumber, triggers...

I also wonder why the library doesnt use "get n0.val" to get the number? Maybe thats the problem?

Would be great if you could have a look at this. Really like the library, but this bug makes it unusable currently

Seithan, 2020-09-23:
Could you please attach the code you are using, in order to see where the bug is?

And please do not say that it is unusable currently, because in this way you are deleting all of my work. 
The library works very well, especially with Hardware Serial, as that is why it was designed.

I have not used the get n0.val command for many reasons. One of them being so I can have the error function.
I could use the get command for the error function and I have already a version with get command, but until now no problems were encountered.

So if you can, send me the code you are using and what MCU the code runs on.

Seithan, 2020-09-24:
I took a look at it yesterday night and unfortunately, you are right. readNumber() is not working with Software Serial.
The error that occurs is that the bytes never reach the Serial for some reason. 

I will attach the Software Serial version with the **get** command for readNumber() and in a future version I will change it and at the Hardware Serial version.

Thank you for alerting me for the bug and what does not work.

[1_0_4_SoftwareSerialV2.zip](https://github.com/Seithan/EasyNextionLibrary/files/5274125/1_0_4_SoftwareSerialV2.zip)

Please let me know if everything works as it should.

Jawomo, 2020-09-24:
Just tested the library and so far everything seems to be working now! 
Thanks for your great support!

Regarding making the library compatible with hardware and software serial you may have a look at this library:
https://github.com/bborncr/nextion

They just create one constructor for HardwareSerial and one for SoftwareSerial and let you switch by commenting out a define.
Maybe thats an option for you.

Will close here later after some more testing

Regards

closed 2020-09-25 by Jawomo