Seithan/EasyNextionLibrary #9: How to read NEXTION special returns with EasyNex ?

joelv1964g, 2020-10-20:
Hi Seithan,
I find your library very simple and very practical to use. She is great!
But I have a question: for an application, I need both custom commands and also tactile feedback from the Nextion (65 xx xx xx FF FF FF) and special feedback from the screen (such FE FF FF FF or 04) for using twfile.
How could these Nextion returns be easily integrated into the EasyNex library?

Thank you in advance for your reply.
Have a good day.
Joel

Seithan, 2020-10-20:
Hello,

For the part of the custom commands, you can see this if you have not already: 
[https://seithan.com/Easy-Nextion-Library/Custom-Protocol/](https://seithan.com/Easy-Nextion-Library/Custom-Protocol/) 

As for the 65 xx xx xx FF FF FF return data, which is the touch event, you can use the trigger() functions instead when something is pressed.
And for the special feedback, FE FF FF FF, the library is not ready to receive those special commands and a new function for Serial reading will need to be written.

Isn't there another way to use the twfile? Although I have not tried it, it does not need transparent mode, I believe. 
And also transparent mode is not reliable because during the waiting time for the specified quantity of bytes, Nextion can not execute any other commands, and may indeed hang if the MCU fails to deliver the number of bytes as specified by the command parameter.

joelv1964g, 2020-10-21:
Hello,
Thank you for your prompt response.
twfile does use Transparent Data Mode. The data must be transmitted in packets (with a header file, which is not clearly stated in the NEXTION documents), and it's very important to listen the Nextion returns after each sending to know if the previous packet was received correctly and send the next one.
I'm going to look at how to incorporate these returns into existing code to use twfile.
Thank you

Joel

Seithan, 2020-10-21:
Hi Joel,

From what I can understand, you have really delved into Nextion.

Unfortunately, these commands, like the twfile, are only used in intelligent models and I cannot help further, as I do not have one in my possesion.
If there is anything else you believe I can help with, I would be glad to.

closed 2020-11-03 by Seithan