Seithan/EasyNextionLibrary #11: Reduce Execution Time for NextionListen()

jesawyers, 2020-10-30:
First, I love your library!!!

One thing I have noticed is that NextionListen() takes almost 1000ms to execute.  Here is the code I'm using in the loop() to calculate the execution time.

```
    int hz1 = millis();
    myNext.NextionListen();
    Serial.println("NextionListen..." + String(millis()-hz));
```
`NextionListen...975`

Is there a way to reduce the execution time for NextionListen?

I'm running on a Arduino Compatable ESP32 at 80MHz

Thanks for any help

jesawyers, 2020-10-30:
I was calculating the NextionListen() incorrectly.

closed 2020-10-30 by jesawyers

Seithan, 2020-10-30:
NextionListen() must be almost 0ms if it has not received any command.
The only thing it mostly does is to check if Serial has a command. If bytes are found in Serial, it will start to read them. The time it will take will only be sending time for the number of bytes which is not much.