Seithan/EasyNextionLibrary #13: Software Serial Pin definition

Rustie0125, 2020-11-04:
Hi
Im trying to use your library but where are you defining your software serial pin allocations/definitions? 
surley it suppose to be in the main example and not hidden in library somewhere? im trying to use it on a NANO that does not have multiple serials so need to use other pins

Seithan, 2020-11-06:
Sorry for the late answer.
Please read the Following comment first.

I strongly recommend to use the Hardware Serial for Nextion and set the other devices with software Serial if needed.

If you still insist on using the Software Serial or AltSoftSerial, I will attach the files for it. 

I am currently working on a safe way to use the Software Serial without issues.

Have in mind that Software Serial or AltSoftSerial can cause problems to any other libraries that could be used in the same project.

Best wishes,
Thanasis

Seithan, 2020-11-06:
Before you decide to use the SoftwareSerial Nextion you must know that:

**The Hardware serial must always be the first choice. 
This is not only my opinion, but every Software Serial library creator recommends it.**

_“No SoftwareSerial approach can rival true HardwareSerial with a real UART, but when no extra serial port is available, hopefully this alternative library can help in cases where the existing libraries have issues.” Paul Stoffregen creator of AltSoftSerial_

Even the AltSoftSerial has, by far, much better performance than other SoftwareSerial libraries. It also has its own limitations and can still cause problems due to interrupts and the use of the hardware Timer1 (16bit). Lose data or interfere with other libraries is the most common.

Nextion is the Human Machine interface for sending commands and receiving data, some of which can be crucial. Nextion is the most critical and most important part in a project, and you must treat it as such by using the most reliable Serial communication, which is the Hardware Serial.

This is the reason that I decided to not include a built-in SoftwareSerial function, which would sometimes be used recklessly. New users on programming might have a hard time finding the source of a problem and they can be confused or frustrated and abort their tries on programming for a problem that was related to bad SoftwareSerial performance and not their code.

As long as you are aware of the problems that can be caused by using any of the Software Serial libraries and you want to use a Software Serial to communicate with Nextion in your project, you can go and read the instructions on how to install the Software Serial in Easy Nextion Library.

If the reason that you need a Software Serial implement is that you can not use the serial monitor for debugging, consider of using the Nextion itself for displaying the debug messages in a Text Box.  

**Different Arduino Serials**

The Arduino has several implementations of serial UART. Not all are equal.

**HardwareSerial** - Hardware serial is by far the best choice, it has the best performance and is the most robust and reliable.
This is always the best solution when available, but it does have a caveat. The Arduino IDE uses the hardware serial to upload to the Arduino so when uploading new sketches, you may need to disconnect whatever is connected to the RX and TX pins.

The Arduino hardware has built-in support for serial communication on pins 0 and 1 (which also goes to the computer via the USB connection). The native serial support happens via a piece of hardware (built into the chip) called a UART. This hardware allows the Atmega chip to receive serial communication even while working on other tasks, as long as there is room in the 64 byte serial buffer.

Hardware serial can:
– Transmit and receive at the same time.
– Work while the Arduino is performing other tasks.
– Never blocked or blocking the code or slow down.
– handle fast baud rates.
More: [https://www.arduino.cc/reference/en/language/functions/communication/serial/](https://www.arduino.cc/reference/en/language/functions/communication/serial/)

**AltSoftSerial** – AltSoftSerial was written by Paul Stoffregen the creator of the Teensy range of boards. It is the best software serial implementation and should be used instead of the default SoftwareSerial where possible.

– Can simultaneously transmit and receive.
– Minimal interference with simultaneous use of HardwareSerial and other libraries.
– Uses the hardware Timer1 (16 bit) and will not work with any libraries which need that timer.
– Disables some PWM pins.
– Can be sensitive to interrupt usage by other libraries or by the user code.
– Fixed pins: pin 8 for receive, pin 9 for transmit.
– The library must be installed from Library manager.
– No support for ESP

More:
[https://www.pjrc.com/teensy/td_libs_AltSoftSerial.html](https://www.pjrc.com/teensy/td_libs_AltSoftSerial.html)
[https://www.arduino.cc/reference/en/libraries/altsoftserial/](https://www.arduino.cc/reference/en/libraries/altsoftserial/)

**SoftwareSerial** (formerly "NewSoftSerial") - NeoSWSerial was written by Slash Devin and it is the next best choice or at least better than the standard SoftwareSerial library. Like SoftwareSerial it can use most of the regular pins but has a limited range of baud rates, 9600 (default), 19200, 31250 (MIDI) and 38400 only. This isn’t really an issue though. The regular software serial has issues when used below 9600 and above 38400 so slower than 9600 and faster than 38400 should not be used.

– Can have multiple instances on almost any pins, but only 1 can be active at a time.
– Can not simultaneously transmit and receive.
– Can interfere with other libraries or HardwareSerial if used at slower baud rates.
– Can be sensitive to interrupt usage by other libraries or by user code.

More:
[http://arduiniana.org/libraries/newsoftserial/](http://arduiniana.org/libraries/newsoftserial/)
[https://www.arduino.cc/en/Reference/SoftwareSerial](https://www.arduino.cc/en/Reference/SoftwareSerial)

**Old SoftwareSerial (SoftwareSerial in Arduino 0023 & earlier) - Very poor performance**

Rustie0125, 2020-11-06:
Hi Thank you for the feedback, The issues are by limiting the use of this library to Hardware serial ports you are limiting the smaller Arduino Nano, ProMini, etc ability to be programmed and debugged via the serial monitor simultaneously. I will try and use the software serial. the issue is when the project is built into a box, you can access the serial wires to remove them before you can upload a new file..

Seithan, 2020-11-06:
Hi
I do not intent in any case to Limit the use of this Library only for MEGA

My goal is to make the most correct and reliable use for the entire Arduino family.

I cannot suggest the use of SoftwareSerial when it is commonly accepted that this solution is not reliable and can cause many problems, like missing bytes and incomplete commands.

Unfortunately, Nextion Displays need a HardwareSerial to work efficient and we must admit it.

To have Nextion at a Hardware Serial is the first priority and we must do anything to accomplish this.

That’s one of the reasons that the official Nextion library comes with a HardwareSerial setup and only for MEGA.

As I have mentioned, the goal is to find the most correct and reliable use for the entire Arduino family and that’s why I strongly recommend using the HardwareSerial for Nextion and set the other devices with software Serial if needed.

This is something that can be done in most cases like:

- Use an FTDI Serial to USB adapter over a SoftwareSerial to Have Serial Monitor for debug.

- Use the Nextion Editor Debug Mode, see the messages on the right down window or use a text box to print them.

- When the project is built into a box, that means that the project is finished, and you do not need to upload a new code version often. In this case you have only to POWER OFF Nextion instead of removing Rx from Arduino.

There is a rule that Nextion must not be powered from the power circuit of Arduino as Nextion needs a lot of power and will bring the power circuit to its very high limits and possible will burn the circuit after a while.

For Arduino the maximum current draw is approx. 400mA on USB and higher if using the DC power jack.

If other devices are connected in parallel with the Vin pin or the 5V pin you must use a diode to avoid 5 volts supply backwards.

When those conditions are met, then you can freely connect the USB cable and upload your code because this way, the only device that it is powered from the USB is Arduino

askkostya, 2020-11-16:
Hello.
Tell me, is it possible to pass the rest command to the display using your library?
Resets the Nextion Device

closed 2020-11-16 by Seithan