Seithan/EasyNextionLibrary #18: Errors handling when writing to incorrect nextion objects

jon-arduino, 2021-01-23:
First - Great library :)
I think I found an issue though.   If you write to an incorrect object it screws things up and you get random behavior like not capturing button presses, etc.
I did a writeNumber("b0",10);    instead of the correct  writeNumber("b0.val",10);
There is no easy way to catch the error, and I might have expected that write would just not happen.
But Nextion display gives and error back "1A FF FF FF" which I think causes your routine to get lost.
It would be great if your library could catch these error codes and call a user routing "Nextion_error()" or something like that in a similar way that you call trigger0() functions.   It would be great if you passed the string that was sent to the nextion that caused the error to this error routine "Nextion_error(String)".   That way the error handler could then help with debug.

Seithan, 2021-01-24:
Hello,
This error caption function will be needed only for debugging, when the code is still being built.
In the final version, there is no use in taking these kind of errors, because you will have corrected them.
There is no need in making the library more complex, and this is one of the library's main goals; To be as simple and easy to use.
I never needed this kind of function in whatever project I have made, and believe me, they are many.
**I always use the debug mode on the Nextion Editor.** It is very simple and it gives all the errors and info you might need. You also do not need a Nextion display to fix the code, which means time wasted by uploading projects via the SD card for a very simple correction.

As for the return codes over serial, I always use bkcmd=0, in the Preinitialize event of the first page, which means no error codes will be sent from Nextion, so that the library does not receive unnecessary data.

jon-arduino, 2021-01-26:
you are correct that if there are no bugs in the code, there is no need to capture errors.   The problem is that over the course of development, there are a lot of bugs.   Often there are still bugs in shipping products or new bugs introduced with later revs.   So having efficient ways to capture bugs and debug efficiently is an important aspect of a library.   If you are pressed for code space, you can turn off these features and save space, but for anysignificant project debbuging support is a necessisty.  I use the debug mode for the nextion editor, but the bugs I have difficult with are with the arduino communication between the display and the arduino.  (using your library).

It would seem that looking for a debug code coming from nextion would be easy to catch.   Just as you support trigger routines today.   If the Nextion generates an error, or you recieve an un-know command, you call a routine called Nextion-Error() or something like that.  This would greatly improve the usefulness of the library.

Seithan, 2021-01-28:
I really believe that there is no need and I do not want to add a debug mode in the library.
The main reason being keeping the simplicity of the library.
I have not seen many libraries with a debug function while running a code.

Nextion's original library has a debug function.
To have a debug function while running a Nextion Display, you will have to use another Serial port for getting the errors or you must print the error messages in a textbox.
Especially the second option cannot happen through the library because every project will have different pages, components etc.
Sometimes I use a tetxbox to print messages from my code for debugging. In the code, I write commands to send messages where I want to make sure that the function has been completed.
Also the use of the library is quite simple that is not affected by changes in Hardware or later revisions. It simply reads the Serial.

Although I am concerened that you are having problems and bugs with communication using Easy Nextion Library and I want to see it closer. If you do not mind, send me your .HMI and .ino files for Nextion and Arduino accordingly, so I can see what the problem is.

If you do not want to post them here, you can use my email seithagta@gmail.com

closed 2021-02-11 by Seithan