Seithan/EasyNextionLibrary #19: input buffer overflow

jon-arduino, 2021-01-23:
I found that if the Nextion generates too much input and the serial input buffer backs up.   It looks like this is not detected properly and the Arduino crashes and reboots.   Library probably writes past the end of an arrray.
Here is how I created the problem.
I put a half second sleep in my main loop - to test for problems.
I then opened a page on nextion with a couple of buttons that call trigger functions trigger0 and trigger1.
Trigger0 and trigger1 just update a text box and change the color of the button.

rapidly hitting the buttons caused trigger calls to be queued up and slowly processed one per half second.
I was impressed that many were queued and that everything kept working.
but with rapid button pressing (10 or twenty quick presses).  The arduino crashed.   I repeaded this several times and it consistently crashes.

I would have expected that button pushes would eventually just get dropped.   Ideally the error would be caught and an error function called to log the error.   But it would be ok to just detect the overflow and drop commands.   

Thanks for the nice library - really helpful as its much nicer to use that others.

Seithan, 2021-01-24:
Hello,
This is the reason that we do not use delay() in our main loop. This can cause many problems, such as the one you mention.
When you use the screen, the maximum presses a human can do is around 5-10 times per second.
This means a command every 100 - 200ms, a time that Arduino can handle easily, as in that time it can execute many other instructions.

By putting a delay(), we are just simply stopping the Arduino and do not let it do anything at all.
This is why the Arduino crashes. 

A good designed code cannot have delay() at all.
Also, the reboot and the crash should not have happened, even if you used delay()
So something else is causing that.

All this are irrelevant with the operation of the library. As you might have seen, in the examples I show some ways on how to write an efficient code. I have examples for how to use the millis() instead of delay(), something found on the internet, where everyone suggests avoiding delay()

I have also examples on how to refresh data on a page, using the last sent data method, so as not to send unnecessary data over Serial. Also, I always recommend the commands that send data over Serial not be inside the loop(), as it will cause buffer overflow, if used without a slow down method, such as millis()

jon-arduino, 2021-01-26:
I don't normally  use delay commands in my code, I was just using the delay to simulate other work the Arduino may need to do.
I think it is unlikely that It will have so much to do that I/O will back up.   But I was testing to see what it does in  the limit.   As stated I expected it would drop button presses.   But instead it crashes.   I think it is probably a relatively easy fix in the lib, but for now I will just work around it by making sure that Arduino does not let things get backed up.

closed 2021-02-11 by Seithan