Seithan/EasyNextionLibrary #20: nextion listen issue

giannaros900, 2021-01-26:
NextionListen function works with 2.4'' but not with 7'' (same code)

Seithan, 2021-01-28:
Hello,
This normally should not happen.
Make sure that you have started a new tft file, where you have set as the device the 7 inch screen
Also make sure that you write **bkcmd=3** at the Preinitialize event of the page

giannaros900, 2021-01-30:
Still the same...

Seithan, 2021-02-12:
I am very sorry for the delayed reply.

I try to find a 7 inch screen, in order to check this out.

I have not been able to find one unfortunately, but from different users on Nextion, that I am in touch, I know very well they use 7 inch screens without problems.

Have you tried the examples of the library?

closed 2021-02-17 by Seithan