Seithan/EasyNextionLibrary #22: HardwareSerial selection

Zefram88, 2021-02-04:
Hi, as in title, i was wondering if there's any possibility to switch to other HardwareSerial ports, i'm using an ESP32 and would be nice to try Hardware Serial 2 (pin 16 and 17) because i'm already using the first serial as monitor for testing on usb.
I'm asking this mainly because i saw a strange behavoiur with a lot of strange characters from ESP to nextion.

Thanks

Seithan, 2021-02-06:
Hello
Try Serial2 in place of Serial
````Cpp
EasyNex myNex(Serial); // Create an object of EasyNex class with the name < myNex > 
                       // Set as parameter the Hardware Serial you are going to use
````

dlaps007, 2021-02-10:
Hello, this method works for me using Serial2 with ESP32.
I use the Serial to print when debugging.

closed 2021-02-17 by Seithan