Seithan/EasyNextionLibrary #23: TriggerX where X > 9 cause kernel panic followed by reboot

RocketBlueAutomation, 2021-02-09:
// Demonstrate crash when triggerxx where xx < 9
// Possible cause: hexadecimal case statements in callTriggers.cpp dont't compile correctly starting on line 67
// -- mikeseiler2 (at) gmail.com

#include "EasyNextionLibrary.h"

EasyNex myNex(Serial2); // Create an object of EasyNex class with the name < myNex > 
                       // Set as parameter the Hardware Serial you are going to use
                       
void setup() {
  myNex.begin(9600); // Begin the object with a baud rate of 9600  // !! change to 115200 on next upload
                     // If no parameter was given in the begin(), the default baud rate of 9600 will be used 
  Serial.begin(115200);
  myNex.writeStr("page page0");   // rename this with the next refactoring
}

void loop() {
  delay(1);  // only read current every few milliseconds
  myNex.NextionListen(); // This function must be called repeatedly to response touch events
                         // from Nextion touch panel. Actually, you should place it in your loop function.
}

void trigger0(){  // OK
  Serial.println(0);
}

void trigger1(){
  Serial.println(1);  // OK
}

void trigger2(){
  Serial.println(2);  // OK
}

void trigger3(){
  Serial.println(3);  // OK
}

void trigger4(){
  Serial.println(4);  // OK
}

void trigger5(){
  Serial.println(5); // not tested
}

void trigger6(){
  Serial.println(6); // not tested
}

void trigger7(){
  Serial.println(7);  // OK
}
void trigger8(){
  Serial.println(8);  // OK
}

void trigger9(){
  Serial.println(9);  // OK
}
/*
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x00000000  PS      : 0x00060f30  A0      : 0x800d1236  A1      : 0x3ffb1f30  
A2      : 0x3ffc00b4  A3      : 0x00000000  A4      : 0x00003df2  A5      : 0x00005980  
A6      : 0x00060f20  A7      : 0x00000000  A8      : 0x800d10e0  A9      : 0x3ffb1f10  
A10     : 0x3f400f5c  A11     : 0x00000000  A12     : 0x3ffb8274  A13     : 0x00000000  
A14     : 0x3ffb82a9  A15     : 0x3ffb0060  SAR     : 0x0000000a  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c28c  LEND    : 0x4000c296  LCOUNT  : 0x00000000  

Backtrace: 0x00000000:0x3ffb1f30 0x400d1233:0x3ffb1f50 0x400d1019:0x3ffb1f70 0x400d0d0b:0x3ffb1f90 0x400d1ab1:0x3ffb1fb0 0x40088215:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac

*/

void trigger10(){
  Serial.println(10);  //CRASH  !!!
}

void trigger11(){      // CRASH !!!
  Serial.println(11);
}

void trigger12(){      // CRASH  !!!
  Serial.println(12);
}

Seithan, 2021-02-09:
Hello
Have you pass the trigger id in HEX format?
 writing in a Nextion Event the command: printh 23 02 54 XX , where XX the id for the triggerXX() in HEX
For trigger10()  printh 23 02 54 0A

RocketBlueAutomation, 2021-02-10:
Thanks for your quick response,  Yes, changing the digits to hex fixed the problem.
Further thanks for your excellent abstraction and simplification for the Nextion interface.

closed 2021-02-11 by Seithan