Seithan/EasyNextionLibrary #24: Unable to read/write values from/to Nextion

dlaps007, 2021-02-11:
First of all like many people posting here many thanks for this super library. It allows many people like me who is not impressed by the Nextion library and forum which is non existant to be able to use the Nextion displays.

I have been using the EasyNextionLibrary with an ESP32 for many months without any issues other then fixing normal development bugs. With all the good documentation I was able to develop a complex application using the library with the trigger() functionality.

I have been able to read/write values from/to variables using the readNumber/writeNum without any problem. Suddenly I can't read/write anymore. When reading a number with the following code: TempToReach = yNex.readNumber("nTempToReach.val"), I get an error. Same thing when trying to write with: myNex.writeNum("nDateYear.val", RTCYear); the value is not updated on the Nextion like it was before.

What is strange is that I know the communication between the Nextion and the ESP32 works because I put Serial.print commands in the ESP32 code and I print when I get to the ESP32 after the Nextion executes the printh 23 02 54 XX.

Any idea what may have cause this functionality to cease functioning after so many months? I tried many small tutorial files with the same issue.

dlaps007, 2021-02-11:
Here is a very basic program that I put to demonstrate the problem.

Arduino code:
`
#include "EasyNextionLibrary.h"             // Include EasyNextionLibrary
#include "SD.h"

EasyNex   myNex(Serial2);                   // Create an object of EasyNex to communicate with the Nextion display using Serial2

void setup() {
Serial.begin(115200);
myNex.begin(9600);                      // Begin the object with a baud rate of 9600
}

void loop() {
  myNex.NextionListen();                  // Make listening to Nextion commands first thing in the loop
}

void trigger1() {                // ***** HEX 01 Triggered from: Manual Settings Screen Post Initialize Event

  Serial.println("In trigger 1");
  String strTest = myNex.readStr("t0.txt");
  Serial.print("test is: ");
  Serial.println(strTest);  
  myNex.writeStr("t1.txt", "From ESP32");  
}`

Picture of the Nextion page and click event of the button to execute code on the ESP32:

[image]

Image of the print debug on the ESP32 to show that Nextion can communicate with it:

[image]

Seithan, 2021-02-12:
Hello,

I am glad that the library has been helpful. :)

It is very weird that the project was working and suddenly stopped.
I cannot find a reason about that.
The `ERROR` fail return value is there on purpose, in case the function fails to read the new value.
The reasons of getting `ERROR`: (from release 1.0.4 and above)
   -  Serial buffer occupied timeout
   -  Waiting bytes have not come to Serial timeout
   - Command start character is not found in Serial timeout
   - The end of the command has not come to Serial
   
The chances of getting a wrong value is one in a million.
You can use this, fail return value, feature in your code, in case you handle sensitive value data, to confirm that you have the right value. 
You can check it with an **`if()`** statement, in which you will ignore the value of `ERROR` and you can run the `readStr()` again or set a safe value or use the last good known value method.

````Cpp
String text = "";
String lastText = "";

text = myNex.readStr("t0.txt");   // We read the value of t0 and store it
    
if(text.equals("ERROR") == false){       // ERROR is the return value if the code fails to read the new value
  lastText = text;
  
} else if(text.equals("ERROR") == true){
    text = lastText;
}
````

To help you debug this problem
- It would be good to download the latest version of the library.
- ESP and Hardware Serial are quite confusing. Try your code on an Arduino
- Try to use Nextion on `Serial` and not on `Serial2`.
- For debugging, use the Nextion Editor in `debug` mode and you can use, as you already did with t1, a textbox to print some debug text there, instead of sending them with Serial print, in order to free up the Serial.

Waiting on your reply :)

dlaps007, 2021-02-14:
Here is where I am.

I downloaded the latest version of the library as your suggestion.
I tried the code on Arduino and couldn't have it to work.
I tried all three serials available on the ESP without success.
I had never tried the debug mode with a USB to TTL adapter as I didn't know it existed. I searched the Web after your suggestion and I found how to make it to work. At the beginning I had a Serial.print in a loop that was printing every time I was in the loop of the ESP32 and after a certain time I noticed an error message in the debug window that there was an buffer overflow. I changed the baud rate of the Nextion and ESP32 to 19200 and the debugging worked and communication between Nextion and the ESP32 was flawless every time I was clicking the "Click" button as per following pictures:

Before the click:
[image]

And after the click:
[image]

And the Serial.pring on ESP32 debugging window:
[image]

So this proves that the communication between the ESP32 and the Nextion is perfect.

After I just disconnected the RX and TX wires from the USB to TTL adapter and connected them to the RX and TX wires of the Nextion without any other modification. I was expecting all to work but nothing works! When I click the button, nothing changes on the Nextion display and the error I'm getting from the beginning appears on the ESP32 debugging window:
[image]

Isn't the Nextion debugging supposed to give the same results once connected to the display? What is very weird is at a certain point everything was working fine and then stopped working after a few trials.

I invested months to learn the Nextion and to come to a certain knowledge and then everything collapses. I'm about to give up and search a new avenue to find a solution similar to the Nextion.

Seithan, 2021-02-14:
Hello,

There are some more displays like Nextion. From what I remember, they are more expensive.
To the matter. First of all, thank you for opening this issue because you are helping me find possible faults in Nextion or in the library.

I will suggest some things that we take for granted, but sometimes we forget.
- Always use a common ground between Arduino/ESP and Nextion
- The default baud rate of a Nextion display is 9600. It is good to write every time in the preinitialize event of the first page that is going to be loaded this: `baud=9600`. This will set the current baud rate to 9600. If you want to change the default baud rate write this: `bauds=115200`
In your project, prefer the 9600
I think this is where your problem is.

Try the above and if there is no success, we must see the following:
- EasyNextionLibrary.cpp  Go to line 81, where you can find the `readStr()` function
In line 91, leave the `ERROR` as it is
In line 114, rename the `ERROR` to `ERROR1`
In line 129, rename the `ERROR` to `ERROR2`
In line 155, rename the `ERROR` to `ERROR3`

Do the following so as to determine in which part the problem is coming from.

You could try to change in your code this line: 
````Cpp 
myNex.writeStr("t1.txt", "From ESP32");
````
to this:
````Cpp 
myNex.writeStr("t1.txt", strTest);
````
This is to have the message stored printed on Nextion.

Also, try to work with `Serial` ONLY. Do not use any other Serial at the same time.

Another thing is to have a logic converter from 3.3V to 5V, as ESP send commands at 3.3V and Nextion uses 5V. This could be sometimes a problem.
because ESPs are special in their own way with the Serials, I am going to send you something else to try after that.

Seithan, 2021-02-14:
Important remove the line from the code
`#include "SD.h"`
As the library may use the same pins with Serial2

dlaps007, 2021-02-15:
AHHHHHHH **Always use a common ground between Arduino/ESP and Nextion... %#@$#%$#$%**

What I didn't remember is that a certain time my Nextion display refused to turn on after loading the code.  My display was connected to the ESP 5V and GND and I read on the Nextion site that it had to be powered by a separate 5V 1A source as the source from the ESP was not providing enough Amp and that could burn the display.

This is when I started to power the display from an independent power source as shown on the Nextion site. They never mentioned a common GND as per the web instructions:

[image]

Powering the display from an external source didn't seem to resolve the problem. I thought my display was dead because not connecting to required specs. I ordered another display (that arrived two weeks later) and after loading the code I had with the same issue. 

After investigating all the possible sources for the problem, I realized that at some point I tried some code in the preinitialize event of the first page that entered in an endless loop and this is what was causing my display to not turn on after loading the code! After changing the code, the display was turning on but this is when the communication problem started. Since there was more than two weeks between the time I started to have problems, I didn't remember the switch I made for powering the display.

What I did after looking at your recommendation is to connect the display directly to my ESP as before and voilà it worked. 
So now I will have to use the external power source but using a common GND and everything should be fine and to the Nextion specs.

I'm so sorry for taking your time for such a basic problem. I just hope this can serve other people that could have the same problem.

This also forced me to learn how to use the debug feature of the Nextion connected to a USB to TTL adapter. Before, I was loading the code to the display every time I was making changes to the display or the ESP code, which was tedious. From now on, I will test everything using the debug and only upload to my display when done.

Again thank you very much for taking the time to help me. I came very close to give up, but by helping my you gave me the tools to figure the problem. Help from Nextion is absolutely NULL and you have to pay if you want them to reply to your questions. Your library is great and the support you give is 5 stars.

Seithan, 2021-02-16:
Hello,

Thank you for kind words. I am really glad when my work is appreciated.

As for the 5 stars, I always try to give the best help possible. Speaking of stars, you could star the library, at the top right of the GitHub page :)

I leave it up to you to close this issue if you do not need this issue anymore :)

dlaps007, 2021-02-16:
Since you kindly solve my problem I will close the issue and did star the library.

closed 2021-02-16 by dlaps007

trulyBent, 2022-08-09:
Seithan;

I've run into the same problem as above. As you suggested, I modified the EasyNextioLibrary.cpp for ERROR, ERROR1, ERROR2, and ERROR3. Now I'm getting ERROR2 each time (line 129 if 0x70 is not detected). 

I lengthened the wait time on line 128 from 100UL to 200UL, but it has made no difference.