Seithan/EasyNextionLibrary #25: Collect sleep information

D-Salmon, 2021-02-17:
Hi,
I've discovered your awesome library and I have a request.
To stop sending commands for refreshing values to Nextion, I'd like to be informed when Nextion is in sleep mode.
I've seen that the display sends 0x86 but I haven't found how to collect it wih your library

ricreis394, 2021-02-17:
I'm interested in this. Subscribing...

Seithan, 2021-02-19:
Hello,

How are you planning to make the Nextion go to sleep mode? With what way?

D-Salmon, 2021-02-19:
Hi Seithan
I only configure in Nextion
thsp=15
thup=1
When the display enters in sleep mode after15 seconds, it sends 86 FF FF FF
It sends 87 FF FF FF when it wakes up
If I could read these informations, I would stop communication beetween Arduino and Nextion for a while

Seithan, 2021-02-20:
Hello
I made some modifications. I will attach the files.
You have to replace those two files in the library's folder and then give it a try.
A new public variable is there
`boolean sleep` that is `true` when a `86 FF FF FF` is received and `false` when `87 FF FF FF`
````Cpp
    if(myNex.sleep == true){
    myNex.writeStr("t1.txt", "sleep = true");
  }else if(myNex.sleep == false){
    myNex.writeStr("t1.txt", "sleep = false");    
  }
````
[EasyNextionLibrary.zip](https://github.com/Seithan/EasyNextionLibrary/files/6015741/EasyNextionLibrary.zip)

Seithan, 2021-02-21:
Hello,

I made some changes in these files. I removed some debug commands that I forgot.

[EasyNextionLibrary.zip](https://github.com/Seithan/EasyNextionLibrary/files/6018456/EasyNextionLibrary.zip)

D-Salmon, 2021-02-24:
Hello Seithan
Fantastic, exactly what I needed.
Thank you so much for your job

Seithan, 2021-02-24:
Have you tried them?

Please give me a feedback if the files are working, so I can change them in a future release.

If you do not need anything else from this issue, close it and give a star to the library on the top right corner :)

D-Salmon, 2021-02-25:
Hi Seithan,
Of course I've tried and it works like a charm.

closed 2021-02-25 by D-Salmon

ricreis394, 2021-03-14:
This feature was already released?

Seithan, 2021-03-17:
Hello @ricreis394,
I have not released this feature yet, as I want to see some other modifications.

Do you need the release now or can you do it with the two attached files?