Seithan/EasyNextionLibrary #26: ESP32 serial communication issue with Nextion

digitalbird01, 2021-02-24:
I am having trouble writing data from ESP32 serial2 to Nextion. I can receive hits from the Nextion but don't seem to be able to write anything is this anything you have come across?

Seithan, 2021-02-24:
Hello
You can check this Issue and pleace let me know if your problem is solved.
#24

digitalbird01, 2021-02-25:
Hi Seithan
It works!!!! Many thanks for the help I have been pulling my hair out for
two days on this basic issue. I was powering the nextion from a separate
supply.
No common earth!  As soon as I powered both the nextion and the ESP32 from
two USB ports on the same system everything worked. Know I can get on
with the hard part.😀

Seithan, 2021-02-25:
Hello,

I believe you do not need this issue no more. You are free to close it, but before you do that, go to the top right corner of the GitHub page and star the library :)

digitalbird01, 2021-02-26:
Awesome, thanks!

closed 2021-02-26 by digitalbird01