Seithan/EasyNextionLibrary #27: Sending string to textbox on nextion

quattroerik, 2021-03-13:
Hello,
first of all thank you for the library. Made my first steps much easier :-)
Second, maybe my problem is not with your library, in that case sorry for troubling you!

Setup: NX3224T024, NodeMCU with ESP8266.
Display is on, Triggers created by Buttons on the Nextion are executed. From the ESP8266 i am connected to my MQTT broker -> communication is up and running both ways.

When i try to read a global variable from the nextion, i only get ERROR.
When i try to write text into a textbox, it only works with a string but not with a variable created. If i send the variable back over MQTT i get the proper value...

Function (callback from MQTT PubSubClient)

void callback(String topic, byte* payload, unsigned int length) {
  String payloadStr;

  //convert payload to String
  for (int i = 0; i < length; i++) {
    payloadStr += (String)payload[i];
  }
  payloadStr = payloadStr.substring(1,payloadStr.length() -1);  //removing the "" from the string
  myNex.writeStr("tempOut.txt", "6.9"); <-- this works
  myNex.writeStr("tempIn.txt", payloadStr); <-- this does not work
}
if i change the two write commands, the other textbox is filled with the "6.9"...

Thanks 
Erik

Seithan, 2021-03-13:
Hello Erik,

The problem for sure is not from the library, as I it works for almost a year without a problem and it is used by many.
However, you are not troubling me. I will be happy to help as much as I can.
The library has been tested and can send variables with the writeStr() function.

You can see this at the ChangePagesAndSendFloatValues example, where at lines 230 and 231, we write this:
````Cpp
    String tempString = String(temperature, 1); // Convert the float value to String, in order to send it to t0 textbox on Nextion
    myNex.writeStr("t0.txt", tempString);       //Write the String value to t0 Textbox component
````
I do not know how long the **payloadStr** is. Nextion cannot take above 22 characters in one command, if I remember correclty.

So try to send a smaller string and see if that helps.

quattroerik, 2021-03-15:
Hi Seithan,

thanks. Uff. Long story, i tried almost everything (display, nodemcu, other code, wrote it new, different library...)
Finally it was the creation of the string itself. Just changed this line

payloadStr = payloadStr.substring(1,payloadStr.length() -1); //removing the "" from the string
to:
String payloadStr = payloadStr.substring(1,payloadStr.length() -1); //removing the "" from the string

and it works.
Beautiful :-)

closed 2021-03-15 by quattroerik