Seithan/EasyNextionLibrary #28: problem with myNex.writeStr("tX.txt+") ?

VicenteYago, 2021-03-22:
Hi,
im using an NX4827T043  and with the following code: 

```{c++}
     myNex.writeStr("t9.txt", "DoW");
     myNex.writeStr("t9.txt+", " hours"); 
     myNex.writeStr("t9.txt+", " mins");
     myNex.writeStr("t9.txt+", " dd/mm/yyyy");     
```
the full text should be displayed, but i obtain this output:

[image]

The full code can be found in : 
https://github.com/VicenteYago/AdvancedWeatherStation/blob/seithanM/src/OW2nextion/OW2nextion.ino  

lines 173-176.

Btw, i want to say that I tested all the available libraries, and this is the best one in quality and documentation.  Good work!

Seithan, 2021-03-23:
Hello ,

Seeing your problem, I concluded that the problem must be at t9 textbox's `txt_maxl` attribute. By default is set to 10 characters length. Change the value to the one that best suits your needs.

Also, You could write the whole command in one line instead of writing it in 4 seperate commands. For sure the way you do it is better, but the string you send is a small one consisting of 25 characters. It is better to seperate and send the big strings with multiple commands. 
You can try to send the whole string with two commands.

If you still experience problems, do not hesitate to write again.

P.S. Your project idea is very good

VicenteYago, 2021-03-23:
thanks now works like a charm!

closed 2021-03-23 by VicenteYago