Seithan/EasyNextionLibrary #32: How to change the picture on nextion?

nodoubtman, 2021-06-06:
Can we change the picture on the nextion display?

And how?

Thanks.
Marc.

Seithan, 2021-06-06:
Hello,
Could you be more specific?
And please read the documentation as you might find what you are looking for.

nodoubtman, 2021-06-06:
Hello, i mean how to change Picture ( p0, p1, p2, etc...)

It's not written in the docs

thanks.
Marc.

Seithan, 2021-06-06:
Hello,
In picture components, there is a numeric attribute "pic", where the value is the id of the picture.
You can use the writeNum() function, and as the documentation says: 
"Description: For writing numbers in numeric attributes in components of any kind (variables, textboxes, etc.)."
[https://github.com/Seithan/EasyNextionLibrary#function-writenum](url)

closed 2021-06-09 by Seithan