Seithan/EasyNextionLibrary #34: FR Reed Touch Coordinates

Terixxx, 2021-09-04:
Thanks for the great work in this library and i have to ask for help. In my project i need to handle touch coordinates enabled with command: sendxy=1
The returned data is: 
67 (02 DA)x (00 D2)y (01 or 00 pressed and released) FF FF FF 
Could you please help me to implement that future in the existing library class.

Seithan, 2021-09-05:
Hello @Terixxx 

Try those two files I attached here and replace the originals with them.

You can call the public variables sendX and sendY by writing 
```Cpp
int x = myNex.sendX;
int y = myNex.sendY;
````
Also, for the Press Event
```Cpp
int event = myNex.pressRelease0x67;
```

[EasyNextionLibrary0x67_0x86_0x87.zip](https://github.com/Seithan/EasyNextionLibrary/files/7112318/EasyNextionLibrary0x67_0x86_0x87.zip)

Terixxx, 2021-09-07:
Hello @Seithan

Code works perfect! Thank you very much. Now I'm workng on getting time between press and release touch events but this will be hard task  as I see.

closed 2021-09-27 by Seithan