Seithan/EasyNextionLibrary #40: Unable to Read Strings Back (readStr())

trulyBent, 2022-08-09:
Seithan;

I've run into the same problem as in issue #24. As you suggested there, I modified the EasyNextioLibrary.cpp for ERROR, ERROR1, ERROR2, and ERROR3 and now I'm getting ERROR2 each time I try to read back a string (line 129, if 0x70 is not detected). The odd thing is that I was able to read them back previously, but I must have changed something unwittingly and messed things up.

I lengthened the wait time on line 128 from 100UL to 200UL, but it made no difference.
I have checked that the commons (GND) are the same for both the Nextion and the MKRZERO board.
I've tried different baud rates, 4800 to 19200, but that doesn't seem to make any difference either.
The signals to/from the Nextion are passing through 3.3V to 5V level shifters.

Thanks for your time in this. Otherwise a great library

trulyBent, 2022-08-09:
BTW, when I compile my code I'm getting these warnings:

C:\Users\User\Documents\Arduino\libraries\Easy_Nextion_Library\src\EasyNextionLibrary.cpp: In member function 'String EasyNex::readStr(String)':
C:\Users\User\Documents\Arduino\libraries\Easy_Nextion_Library\src\EasyNextionLibrary.cpp:144:48: warning: comparison is always false due to limited range of data type [-Wtype-limits]
if(_tempChar == 0xFF || _tempChar == 0xFFFFFFFF){  // If the read byte is the end command byte, 
__________________________________________^
C:\Users\User\Documents\Arduino\libraries\Easy_Nextion_Library\src\EasyNextionLibrary.cpp: In member function 'uint32_t EasyNex::readNumber(String)':
C:\Users\User\Documents\Arduino\libraries\Easy_Nextion_Library\src\EasyNextionLibrary.cpp:245:46: warning: comparison is always false due to limited range of data type [-Wtype-limits]
if(_tempChar == 0xFF || _tempChar == 0xFFFFFFFF){  // If the read byte is the end command byte,
__________________________________________^

Just thought you might want to see that

Seithan, 2022-09-24:
try to replace 
````Cpp
if(_tempChar == 0xFF || _tempChar == 0xFFFFFFFF){  // If the read byte is the end command byte, 
````
with
 ````Cpp
  if(_tempChar == 0xFF || _tempChar == 0x100){  // If the read byte is the end command byte,
````
in 144 and 245 lines at : src/EasyNextionLibrary.cpp

closed 2022-10-04 by Seithan