Seithan/EasyNextionLibrary #45: Intermittent triggerX() detection and currentPage

PeterBowater, 2022-12-29:
Hi!

Hope you can help, I've been working on a digital dash project and came across your library which makes things WAAAY cleaner for my use. 

Firstly, a quick summary of the hardware/software:

- Arduino UNO R3
- Nextion 7" Intelligent series screen NX8048P070_011
- Nextion editor v 1.65.0

Firstly, I was having trouble getting the currentPage value, and when trying to get it using the global dp variable it was REALLY slow, so abandoned that and went looking for a different solution.

My idea was to trigger a function by getting the Nextion to send a hex string as documented such as : 

`printh 23 02 54 01` 

as per:

[image]

01 for page 1, 02 for page 2 on the button touch release event. That appears to work fine in the Nextion simulator/debug, and *sometimes* it works OK when changing page, the Arduino picks it up and will set the global Arduino Int to the current page, which I push out to the Nextion for testing inside the loop().

Sketch as an example:
```

#include "EasyNextionLibrary.h"

EasyNex myNex(Serial);

int currentPage=1;

void setup() {
  myNex.begin(115200);  
  delay(500);  // This delay is just in case the nextion display didn't start yet, to be sure it will receive the following command
}

void loop() {  
  myNex.NextionListen();
  myNex.writeStr("gearposition.txt", String(currentPage)); // these two fields are extant in the dash, just using for debug here
  myNex.writeNum("pageNum.val", currentPage);
}

void trigger1(){
  currentPage=1;
}
void trigger2(){
  currentPage=2;
}
void trigger3(){
  currentPage=3;
}

```

I've reduced the code to a minimum above (and have tried this here as well) with the same results, it doesn't always update the page number, I'll sometimes end up with page three showing '1' or '2', page one showing '2' or '3' etc.

Am I doing something wrong? Any ideas?

Thanks very much!

Peter.

PeterBowater, 2022-12-29:
Realised I ought to add why I was having trouble with myNex.currentPageId, using the same code above, replacing the writeStr/writeNum values with:

`myNex.currentPageId`

Results in a 0 every time.

PeterBowater, 2022-12-29:
Also after some further experimentation:

`currentPage = myNex.readNumber("dp");
  if(currentPage != lastCurrentPage){
    myNex.writeStr("gearposition.txt", String(currentPage));
    myNex.writeNum("pageNum.val", currentPage);
    lastCurrentPage = currentPage;
  }`

Results in alternating -1 and actual page number. Is -1 a default return if undefined/not completed yet?

Seithan, 2023-01-16:
https://github.com/Seithan/EasyNextionLibrary#library-public-variables

closed 2023-02-06 by Seithan