Seithan/EasyNextionLibrary #47: Using Serial2 on ESP32 with redefined pins

laserir, 2023-02-13:
Hi there,

I need to connect to redefined Serial2 pins. Standard-pins would be 16/17 for UART2, but I need to be using 5/14. My attempt is as follows, but doesn't seem to work... Any ideas? :)

```
#include "EasyNextionLibrary.h"
#define RXD2 5
#define TXD2 14

EasyNex nextion(Serial2);

void setup() {
  Serial.begin(115200); //USB
  Serial2.begin(9600, SERIAL_8N1, RXD2, TXD2); //Nextion
  nextion.begin(9600);
}
```

Seithan, 2023-02-14:
The easy way
go to esp32 library HardwareSerial.cpp and ghange the value on Line 80 and 88
``` cpp
80 #define RX2 16
88 #define TX2 17
```
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/HardwareSerial.cpp

mphohio, 2023-02-19:
Seithan, I have the same issue - I need to assign the RX/TX pins for Serial2.  When I try this I get a error from the library.  I looked at the hardwareSerial.cpp file on my computer and I did not see anywhere to assign RX/TX pins.

Seithan, 2023-02-21:
Hello,

The answer I gave above is ONLY for the ESP32 library. You cannot do the same for Arduino.

You have to use the default Serial pins

laserir, 2023-02-23:
Possible solution...
```
#include "EasyNextionLibrary.h"
#define RXD2 5
#define TXD2 14

EasyNex nextion(Serial2);

void setup() {
  Serial2.begin(9600, SERIAL_8N1, RXD2, TXD2); //Nextion
  nextion.begin(9600);
}
```
Then comment out line 25 in **EasyNextionLibrary.cpp**:
`// _serial->begin(baud);`

This way the initialization of Serial2 is done manually before calling EasyNex's .begin, which lets you choose the RX, TX pins. Also, you keep the correct HardwareSerial settings for other projects or within ESP board definition updates.

Cant try at the moment, but should work (?)

mphohio, 2023-02-23:
laserir, thank you for the response.  I stumbled on this approach by accident and now have my ESP32-S3 working with the Nextion display.