Seithan/EasyNextionLibrary #48: NEXTION is not displaying the data sent to it

mphohio, 2023-02-19:
I am working on a project using an ESP32-S3 and Nextion display.  I found your library and felt is was a perfect fit for my minimal coding skills.  Everything works when using the NEXTION simulator, but when I interface with the display nothing I send gets displayed.  
I submitted a detailed explanation in a request for help on the arduino forum and was wondering if you can talk a look and LMK if you have any advice/suggestion on how to fix my problem.  The details start with post #3.  The link is:
https://forum.arduino.cc/t/esp32-s3-and-nextion-7-intelligent-display/1091880/

Seithan, 2023-02-21:
Hello @mphohio 

The problem you are experiencing is that you do not have a **common ground** between Nextion and ESP

mphohio, 2023-02-23:
Not the problem, but the issue has been resolved.  Thanks!

closed 2023-02-23 by mphohio

Giorgiofossati, 2023-03-23:
> Not the problem, but the issue has been resolved. Thanks!

Hello, @mphohio how you solved your problem?

mphohio, 2023-03-23:
@Giorgiofossati 
My problem was when using the ESP32-S3 I needed to assign the Serial Rx & Tx pins in the begin statement, but the Easy Nextion Library would throw an error when I added anything other than the baud rate.  The solution was to assign myNex to Serial2, then in Setup I started Serial2 with the pin assignments and then started myNex (see below).

`EasyNex myNex(Serial2);

(in Setup)
Serial2.begin(9600, SERIAL_8N1, RXp2, TXp2);
myNex.begin(9600);` 

I am not a programmer so I was not aware that you could start a Serial port and then assign it to an object.  

Mike