Seithan/EasyNextionLibrary #50: Invalid reading from Nextion

s56jsc, 2023-03-05:
Hi
I try to read data from display. If I'm on main page it work OK
`xx = myNex.readNumber("popravi_prog.n11m.val");
    Serial.println(xx);  `
but if I'm on some other page I got an "xxx" number. 
n11m is mark as global. Also in nextion debuger work OK.

Seithan, 2023-03-09:
https://github.com/Seithan/EasyNextionLibrary#usefull-tips

Use `writeStr()` to change the text of a String variable.
Example: `myNex.writeStr("va0.txt", "Hello World");` For this to happen, the variables you want to read/write must be at the page you are currently on.
Otherwise, if the variables are of global scope, you will need to use a prefix with the page name that the variables are at.
Example:
`myNex.readNumber("page0.va0.val"); // If the variable is at page0`
**The same goes for the other functions as well.**

If it works fine on nextion debuger, use a common ground between Arduino and Nextion display.
Pay attention on that Serial.println(xx); command is not using the same Serial with Nextion.

s56jsc, 2023-03-12:
Thanks for your replay.
I'm using same ground and communication works. Also I use serial1 for nextion so this cannot be problem.
But like I say it doesn't works. 
If I'm on main screen (page0) I got correct response, but if I'm on some other page (non page1, page2, ...) it doesn't work. I got value 255
On naxtion debugger it work OK I got always correct number no mater from which page I call this function.

Seithan, 2023-03-18:
try to use standard prefix for Nextion attributes (page name, object name) like page0 instead of popravi_prog and n11 for n11m

s56jsc, 2023-03-19:
Strange, but now works!
Thanks!

closed 2023-03-19 by s56jsc