Seithan/EasyNextionLibrary #52: Nextion & Arduino UNO don't communicate on Loop

Giorgiofossati, 2023-03-23:
Hello, my problem is when i connect arduino uno and nextion they don't comunicate ( ithink)
the strange fact is that i write.string on the setup and the value is displayed...
but when i press button nothing change ( on touch release event i printh 23 02 54 00)

1- the ground is in common (otherwise the value on startup won't be send and showed)
2- if i use the nextion simulator everything work fine ( the MCU respond and the nextion update the txt)
3- baudrate is the same
4- Rx led on arduino is always on ( ithink is right)

could be an hardware propblem of the nextion which can't send command on serial?
thanks for the help
here the code:
#include "EasyNextionLibrary.h"
EasyNex HMI(Serial);

float spd;

void setup() {
  // put your setup code here, to run once:
  //Serial.begin(9600);
  HMI.begin(9600);
  spd = 20;
  String Spd = String(spd, 1);
  HMI.writeStr("t0.txt", Spd);
}

void loop() {
  // put your main code here, to run repeatedly:
  HMI.NextionListen();
  
}

void trigger0(){
  //spd+=0.1;
  spd=21;
  String Spd = String(spd, 1); // Convert the float value to String, with 1 decimal place
  HMI.writeStr("t0.txt", Spd);
}

void trigger1(){
  //spd+=0.1;
  spd=22;
  String Spd = String(spd, 1); // Convert the float value to String, with 1 decimal place
  HMI.writeStr("t0.txt", Spd);
}

void trigger2(){
  //spd+=0.1;
  spd=23;
  String Spd = String(spd, 1); // Convert the float value to String, with 1 decimal place
  HMI.writeStr("t0.txt", Spd);
}

void trigger3(){
  //spd+=0.1;
  spd=24;
  String Spd = String(spd, 1); // Convert the float value to String, with 1 decimal place
  HMI.writeStr("t0.txt", Spd);
}

lenkaiser, 2023-03-24:
Is the Arduino Uno connected to your PC with USB? I've noticed that I have issues with nextion devices if they're. Disconnect the USB and power is separately and it works.

Giorgiofossati, 2023-03-27:
> Have you confirmed that the system works with the code example? I too struggled for a couple of days tbh.
> 
> Edit: I just opened a ticket that sometimes I don't see that the trigger method is called: #53

I tried with with two examble code:
1- **ReadAndWriteNumber** wich read the values fron nextion -> Didnt work
2- **WriteTextAndCommands**  wich send commands to the nextion -> this one work perfectly
So the problem is with the trigger method of the library or with the Rx of the nextion???

lenkaiser, 2023-03-27:
Sending stuff to the Nextion seems to work perfectly but receiving from it doesn't. But also sending stuff isn't 100% accurate. I think it sometimes sends/receives a part of the data and then continues to the next loop. That's why sending a number works but sending a string or receiving the command doesn't.

Seithan, 2023-04-11:
try to rename the Spd String to tempSpd

```
String tempSpd = String(spd, 1); // Convert the float value to String, with 1 decimal place
HMI.writeStr("t0.txt", tempSpd);
```

Giorgiofossati, 2023-04-11:
> Try to rename the Spd String to tempSpd
> 
> ```
> String tempSpd = String(spd, 1); // Convert the float value to String, with 1 decimal place
> HMI.writeStr("t0.txt", tempSpd);
> ```
> 
> Try to use a not float for spd in order to check if works

Hello, i will try that...but the problem persist also with your examples code:

I tried with with two examble code:
1- ReadAndWriteNumber wich read the values fron nextion -> Didnt work
2- WriteTextAndCommands wich send commands to the nextion -> this one work perfectly
So the problem is with the trigger method of the library or with the Rx of the nextion???

Seithan, 2023-04-11:
Unfortunately, the activation method and library have been tested by me and many other users with no problems

Giorgiofossati, 2023-04-12:
> Unfortunately, the activation method and library have been tested by me and many other users with no problems

So the problem could be be on my nextion device, a fulty one maybe?
I'm using the discovery series, you tested the examples on the basic series ?
meanwhile thank you !

closed 2023-04-30 by Seithan