Seithan/EasyNextionLibrary #53: Trigger methods sometimes not called

lenkaiser, 2023-03-24:
I tried the example code but it looks like sometimes the trigger method isn't called. I'm running this on an Arduino UNO so I have no way of debugging the code to see if the input command is missed. Shouldn't there be a confirmation loop that the received command is handled by clearing the last received command?

`#include "EasyNextionLibrary.h"

EasyNex myObject(Serial);
void setup(){
  myObject.begin(9600);
  pinMode(LED_BUILTIN, OUTPUT);
}
void loop(){
  myObject.NextionListen();
}
void trigger0(){
  digitalWrite(LED_BUILTIN, !digitalRead(LED_BUILTIN));
}`

Seithan, 2023-04-11:
Commands are not saved. if a command fails then it will give an error message (readNumber and readString).
The library is checked and usually errors come from the user side, one possible reason is when the pin contacts are loose.

lenkaiser, 2023-04-11:
I’m not sure what the issue is/was but switching to NextionX2 immediatley fixed my issues with touches.

closed 2023-04-11 by Seithan