Seithan/EasyNextionLibrary #54: Read still 777777

puhycz, 2023-04-05:
I have HMI with page0 and page1. Writing data is ok, nextion lissen and trigger works perfectly, but i can't read any variable. I try any variables, local, global but still return 777777:

`myNex.writeStr("page 1")
Serial.println(myNex.readNumber("n0.val"));
`

I think reading works when nextionlisten works on the touch pad. Why readings of variables still return 777777?
Thanks for any ideas.

Platform: ESP32/platformio/arduino framework

Seithan, 2023-04-11:
https://github.com/Seithan/EasyNextionLibrary#function-documentation

NOTE: As these commands are using the Serial port to read and write, it is more preferred not to run them in the loop() without delay(); or some other method of not running them with the frequency of the loop and use them only when it is needed. Using them in a loop, a delay in the loop can be noticed, especially when reading from the Serial. A Serial buffer overflow can also be caused.

18Markus1984, 2023-04-27:
> I have HMI with page0 and page1. Writing data is ok, nextion lissen and trigger works perfectly, but i can't read any variable. I try any variables, local, global but still return 777777:

I have the same issue, I didn't get the controller to send any data. I tried a lot of different configurations by sending the print at different times. The Data always is 777777. I also checked if I can only read the data while I'm on the same page but this also did work. Maybe we can find a solution. I'm using an NodeMCU ESP8266 but I will try it on my ESP32 S2 Mini and the Arduino Uno.

puhycz, 2023-04-28:
I tested it on ESP32. Still read 777777. I tested 4800 - 115200 baudrate. And still the same. 777777. Write is ok, trigger is ok, but when I need read values, return 777777. I use HW Serial2 on ESP32. I try on 3 different ESP32 board. Still same, 777777.

18Markus1984, 2023-04-28:
> I tested it on ESP32. Still read 777777. I tested 4800 - 115200 baudrate. And still the same. 777777. Write is ok, trigger is ok, but when I need read values, return 777777. I use HW Serial2 on ESP32. I try on 3 different ESP32 board. Still same, 777777. 

What if we can read the data on a seperate SoftwareSerial connection we switch to while the trigger is recogniced? And send the data as a printh because the next data char have to be the data?

18Markus1984, 2023-04-30:
Why do closed the issue? Do you have a solution? I really want to get your library working for my university project. It would be super kind to get an answer.

puhycz, 2023-09-21:
So, for example, in this HMI is set global variable va1: [](https://puhy.cz/temp/hmiFinal.HMI) . How read this variable? myNex.readNumber("va1.val")); not function. Still reads 777777. 

Thanks.

Grandpa-G, 2023-10-06:
Any solution yet? I am trying to read rtc0 in setup with 500 delay after. I have tried other variables with the same result - all 7.

Grandpa-G, 2023-10-07:
I belive I have found the problem and have a solution. At line 245 in EasyNextionLibrary.cpp, the checking of the incoming data for the end of data flag(s) is incorrect. The code is looking for either FF or FFFFFF. I have it is FFFFFF which matches. However the code wants 3 bytes of code before setting the _endOfCommandFound flag. When FFFFFF comes in, it is a end of command flag, but  all at once. A solution is to make 2 seperate if statements so the FFFFFF will immediately set the _endOfCommandFound flag and break. I have done limited testing, but it work for me. Thus the new code looks like:

				if(_tempChar == 0xFFFFFFFF){  // If the read byte is the end command byte, 
					_endOfCommandFound = true;  // If the counter is equal to 3, we have the end command
					break;
				}
				if(_tempChar == 0xFF){  // If the read byte is the end command byte,

Seithan, 2023-10-10:
> So, for example, in this HMI is set global variable va1: [HMI](https://puhy.cz/temp/hmiFinal.HMI) . How read this variable? myNex.readNumber("va1.val")); not function. Still reads 777777.
> 
> Thanks.

https://github.com/Seithan/EasyNextionLibrary#usefull-tips

_be at the page you are currently on.
Otherwise, if the variables are of global scope, you will need to use a prefix with the page name that the variables are at.
Example:
myNex.readNumber("page0.va0.val"); // If the variable is at page0
The same goes for the other functions as well._

Always use a prefix with the page name for global variables

Grandpa-G, 2023-10-12:
Something strange is happening. After some time I don't get any triggers to fire. The same code works some of the time and not others.

Seithan, 2023-10-14:
> I belive I have found the problem and have a solution. At line 245 in EasyNextionLibrary.cpp, the checking of the incoming data for the end of data flag(s) is incorrect. The code is looking for [...]

@Grandpa-G 

I think this is not a permanent solution. If you want to try something, consider trying the following:

In line 178, change the `char _tempChar;` to `byte _tempChar;`

Let me know if this works better

Grandpa-G, 2023-10-14:
I will try it but I am not sure how that will help.

I am having more problems with getNumber. I am almost always get 777777. Also there are many times when the trigger doesn't fire. I am at the end of the project deadline so I have to get this working.

Seithan, 2023-10-16:
@Grandpa-G 

If you want, you can send me the .ino and .HMI files of your project in order to have a closer look at it and be of assistance

cattledogGH, 2023-11-11:
I had issues with 777777 from .readNumber( ) statements in trigger events when I had page switching code in the touch release event from the button press which sent the trigger command to the Arduino. I moved the page switching code into the Trigger function and the 777777 errors went away.