Seithan/EasyNextionLibrary #61: ESP32 reboots

puhycz, 2023-09-21:
Hello,

I use [EasyNextionLibrary](https://github.com/Seithan/EasyNextionLibrary) for connect ESP32 with Nextion display. It crashes when I press touch buttons. One or two press is ok, but many presses or quick press again crash. I call myNex.NextionListen(); every 50ms.

Thanks, Jan.

Debug trace:

Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x00000000  PS      : 0x00060130  A0      : 0x800fd058  A1      : 0x3ffd7e30
A2      : 0x3ffc88d8  A3      : 0x0001303d  A4      : 0x00013080  A5      : 0x00060123  
A6      : 0x00060120  A7      : 0x00000001  A8      : 0x800fd07c  A9      : 0x00000054
A10     : 0x00000001  A11     : 0x3ffd7e2f  A12     : 0x00000001  A13     : 0x10624dd3  
A14     : 0x00000001  A15     : 0x003fffff  SAR     : 0x0000000a  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x40090f84  LEND    : 0x40090f8e  LCOUNT  : 0x00000000

Backtrace: 0xfffffffd:0x3ffd7e30 0x400fd055:0x3ffd7e50 0x400dd11d:0x3ffd7e70

  #0  0xfffffffd:0x3ffd7e30 in ?? ??:0
  #1  0x400fd055:0x3ffd7e50 in EasyNex::NextionListen() at lib/Easy Nextion Library/src/EasyNextionLibrary.cpp:340
  #2  0x400dd11d:0x3ffd7e70 in nextionUpdate(void*) at src/thread_nextionUpdate.cpp:348

ELF file SHA256: e10a9f359f05384c

E (6371) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ets Jun  8 2016 00:22:57

`void nextionUpdate(void* pvParameters)
{
    for (;;) {        
        nexChange=0;
        myNex.NextionListen();        
        cntNextion++;
        if (cntNextion == 1) {            
            if (getNexBl()) {
                myNex.writeNum("dim", MAX_NEXTION_BACKLIGHT);
            } else {
                if (nexBlTimer > 0) {
                    int dimmerShadow = map(nexBlTimer, 0, MAX_NEXTION_BACKLIGHT_TIMER, 0, MAX_NEXTION_BACKLIGHT);
                    myNex.writeNum("dim", dimmerShadow);
                } else {
                    myNex.writeNum("dim", 0);
                    nexBlTimer = 0;
                }
                nexBlTimer--;
            }
        }       

        if (cntNextion == 2&&nexChange!=11) {
            myNex.writeNum("t1.bco", thermostat_getD() ? 64512 : 65535); // doma
            myNex.writeNum("t3.bco", thermostat_getK() ? 64512 : 65535); // kancelar
            myNex.writeStr("t4.txt", String(temperature_get(ID_CIDLA_REG), 2) + String(" °C")); // doma
            myNex.writeStr("t5.txt", String(temperature_get(ID_CIDLA_EKV), 2) + String(" °C")); // venku
            myNex.writeStr("t6.txt", String(temperature_get(ID_CIDLA_KANCELAR), 2) + String(" °C")); // kamcelar            
        }
        if (cntNextion == 3&&nexChange!=11) {
            myNex.writeStr("t11.txt", String(getAqIn()) + " AQI"); // AQI IN
            myNex.writeStr("t10.txt", String(getAqOut()) + " AQI"); // AQI OUT
            myNex.writeNum("t11.bco", aqiIndexColor(getAqIn())); // AQi IN COLOR
            myNex.writeNum("t10.bco", aqiIndexColor(getAqOut())); // AQi OUT COLOR
            myNex.writeStr("t13.txt", getWindString());
            myNex.writeStr("t14.txt", (String(getSpotreba()) + String("W")));
            myNex.writeStr("t15.txt", FVEPretok() ? String("je přebytek") : String("není přebytek"));
            myNex.writeNum("t15.bco", FVEPretok() ? 2016 : 63488);            
        }
        if (cntNextion == 4||nexChange==11) {            
            myNex.writeStr("t16.txt", String(thermostat_get_temperatureD()) + String(" °C"));
[... 152 lines cut ...]
    nexChange = 11;
    float tempTmp = thermostat_get_temperatureK() + 0.5;
    thermostat_setK(tempTmp);
    // myNex.writeStr("t18.txt", String(tempTmp) + String(" °C"));
    // mySerialPrintLn(F("TK+"));
}

void trigger12()
{

    /* Create a button on Nextion
     * Write in the Touch Release Event of the button
     * this command:    printh 23 02 54 0C
     * Every time the button is pressed, the trigger0() function will run
     * and the code inside will be executed once
     */

    nexBlTimer = MAX_NEXTION_BACKLIGHT_TIMER;
    nexChange = 11;
    float tempTmp = thermostat_get_temperatureK() - 0.5;
    thermostat_setK(tempTmp);

    // myNex.writeStr("t18.txt", String(tempTmp) + String(" °C"));
    // mySerialPrintLn(F("TK-"));
}`

atoughram, 2023-11-06:
Also have reboot issues, but triggered differently.  Using trigger functions to read buttons, trigger0() through trigger3() work well, but trigger4() causes a reboot on ESP32.  

Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x00000000  PS      : 0x00060e30  A0      : 0x800d2572  A1      : 0x3ffc53a0  
A2      : 0x3ffc1c1c  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x10624dd3  
A6      : 0x3ffbdbc8  A7      : 0x00000000  A8      : 0x800d25d0  A9      : 0x3ffc5370  
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x3ffbb170  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x40085d8c  LEND    : 0x40085d96  LCOUNT  : 0x00000000  

Backtrace: 0xfffffffd:0x3ffc53a0 0x400d256f:0x3ffc53c0 0x400d2531:0x3ffc53e0 0x400d199e:0x3ffc5400 0x400d4a75:0x3ffc5420

ELF file SHA256: dab56bfc02cec773

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4

Seithan, 2023-11-10:
Hello,
Please try changing this file in the library's source files
[trigger.zip](https://github.com/Seithan/EasyNextionLibrary/files/13323073/trigger.zip)

puhycz, 2023-12-09:
Yes, I tested it and it is ok. Many thaks.

challo2018, 2025-08-04:
i had similar crash issue. fixed with changed "trigger" file from @Seithan 

> Hello, Please try changing this file in the library's source files [trigger.zip](https://github.com/Seithan/EasyNextionLibrary/files/13323073/trigger.zip)