Seithan/EasyNextionLibrary #62: Read/Write EEProm

Grandpa-G, 2023-09-30:
Can you give and example of how to read and write an integer into eeprom? I would prefer the value come back to the arduino to process but I will take anything.

Seithan, 2023-09-30:
Hello @Grandpa-G ,

This is not a library issue. Save the value you want into a variable and then store it to EEPROM with the `epprom.write` or `epprom.update`. More details [here](https://docs.arduino.cc/learn/programming/eeprom-guide)

If your value exceeds 255, see [here](https://forum.arduino.cc/t/writing-numbers-bigger-than-1-byte-to-eeprom/56631/5) how to do that

Grandpa-G, 2023-09-30:
The first link seems to talk about an eeprom in the Arduino. I am asking about how to save in the Nextion eeprom? I see the wepo and repo Nextion commands, but I don't understand how to use your library to use them I think I understand writing with the writeStr function, but how do I read it back, readStr? I just get a value of ERROR.

Grandpa-G, 2023-09-30:
Maybe I will switch to Arduino eeprom. Thanks.

Seithan, 2023-10-01:
It is better to store the values that you use for calculations on Arduino and then display the risults on Nextion

closed 2023-10-01 by Seithan