Seithan/EasyNextionLibrary #63: Reading from dual-state button

hayderismael, 2023-10-06:
I am so grateful for all the efforts put into developing this library. 

I use
 `myNex.readNumber`

to read the status of the dual-state button, for example

```
int sw1 = myNex.readNumber("mpid.msv1.val");
 digitalWrite(Rly1, sv1 == 1 ? HIGH : LOW);
```
It is okay when reading one or two buttons, but when there are many buttons, in my case 16 ones, the loop becomes slow and sometimes when hit sw1, sw5 or any other one turn on for a moment.  
Would you please advice what is the best way to read the status of the dual-state button? 

Thank you

Rocket-Rod, 2023-10-10:
Hello hayderismael,
Is your code meant to read;

int sw1 = myNex.readNumber("mpid.msv1.val");
 digitalWrite(Rly1, **_sw1_** == 1 ? HIGH : LOW);

or

int **_sv1_** = myNex.readNumber("mpid.msv1.val");
 digitalWrite(Rly1, sv1 == 1 ? HIGH : LOW);

Rocket-Rod, 2023-10-10:
If all of your **_myNex.readNumber()_** are in the loop you will experience delays as the documentation points out.
My suggestion would be to use the predefined triggers to only call your code when needed, leaving the loop to just listen for the Events.
Example:
In your Nextion project for the Dual-state button, write in the Touch Press Event or Touch Release Event (the choice is yours)
printh 23 02 54 00

Now in the Arduino code after the loop (which is doing the listening) write something like the following:
void trigger0() { //Event: printh 23 02 54 00 - Touch Press Event for Dual-state button msv1 on Page mpid
int sv1 = myNex.readNumber("mpid.msv1.val");
digitalWrite(Rly1, sv1 == 1 ? HIGH : LOW);
}

You can do this for 50 pre-defined events. It is explained in the documentation how to increase this number if required.

closed 2023-11-10 by Seithan