Seithan/EasyNextionLibrary #68: Send Component ID blocks Triggers

cattledogGH, 2024-11-18:
When the Send Component ID box is checked for a Touch Press Event or Touch Release Event which is a trigger command (e.g. printh 23 02 54 00) the trigger function is not recognized by an Arduino.

Tested with the library example TriggerCode.ino  and Trigger.HMI.  Checking and unchecking the Send Component ID box  for the button events will determine whether the Arduino responds to the trigger or not.

This behaviour should be documented, or else the .NextionListen() function and command match should be modified to accommodate the component id being sent before the trigger bytes.

eforekim, 2025-02-02:
I am having a similar issue though I am not even sending component id. I found that leaving page exit event empty blocks the next command sent, not only triggers but anything using printh.  also, triggers are almost never working correctly. what is that about?

cattledogGH, 2025-02-03:
@eforekim 
I don't understand the details of your issue, especially "triggers are almost never working correctly". I have any HMI files with empty page exit events, so i don't understand as well.

My suggestion is to take this discussion to the Arduino forum. I think it will be easier describe your environment in detail, to show your code, and attach HMI files.

If post on the Arduino forum , I will pick up your issue there.